sair-lab/AirSLAM #185: 你好，roslaunch后是会出现rviz显示轨迹吗，我在docker内没有任何可视化显示，但是轨迹txt有显示

Ddd195, 2025-02-04:


Ddd195, 2025-02-04:
加了<node name="rviz" pkg="rviz" type="rviz" args="-d $(find air_vo)/configs/euroc.rviz" output="screen" />这句后euroc数据集可以可视化显示了，但是另外两个数据集没有对应的rviz文件，应该如何可视化呢

xukuanHIT, 2025-02-16:
你好，不知道你启动的是哪个launch没显示的呢？提供的launch里面应该都默认开启了rviz，类似[这行](https://github.com/sair-lab/AirSLAM/blob/2b825166b05b351fc708c70cee3fc8b5a5b380e9/launch/visual_odometry/vo_oivio.launch#L18)。

Ddd195, 2025-02-26:
> 你好，不知道你启动的是哪个launch没显示的呢？提供的launch里面应该都默认开启了rviz，类似[这行](https://github.com/sair-lab/AirSLAM/blob/2b825166b05b351fc708c70cee3fc8b5a5b380e9/launch/visual_odometry/vo_oivio.launch#L18)。

已经解决了，请问umavi数据集是怎么评估的呢，我对umavi数据集的imu0_trajectory.csv时间戳变成了秒，然后去掉逗号
evo_ape tum -a，umavi数据集的效果不及论文效果，并且十分不稳定，是我的评测方式有问题吗？

xukuanHIT, 2025-02-27:
你好，我们也是用的evo评估的。你跑的结果是每次轨迹差很多吗？还是每次轨迹类似，但是rmse算的结果不稳定？可以用evo先对比下两次结果轨迹的差别

Ddd195, 2025-02-28:
> 你好，我们也是用的evo评估的。你跑的结果是每次轨迹差很多吗？还是每次轨迹类似，但是rmse算的结果不稳定？可以用evo先对比下两次结果轨迹的差别
每次轨迹看起来相差并不大，但是evo评测的时候rmse相差较大。我使用了evo对两次轨迹进行evo_ape tum -a 评测，rmse接近1。另外，请问"真值只有开始和结尾处的部分"是什么意思呢。应该如何正确评估输出的traj.txt和真值imu0_trajectory.csv，非常感谢您

Ddd195, 2025-02-28:
> 你好，我们也是用的evo评估的。你跑的结果是每次轨迹差很多吗？还是每次轨迹类似，但是rmse算的结果不稳定？可以用evo先对比下两次结果轨迹的差别

但是如果我加上--correct_scale选项，rmse将会变得非常低，相比论文的结果0.5236会低一半，来到了0.285309，这样是正确的吗？

xukuanHIT, 2025-02-28:
UMA-VI只有初始部分和结尾部分用SFM和标定板获取的真值，中间轨迹部分是没有真值的，所以在评估时不能对齐尺度，也不能用于评估带回环检测的算法。结果不稳定有可能是每次轨迹的微小改变会影响evo对齐的结果，造成了得到的RMSE差很多。另外，可以忽略最后的"terminate called without an active exception"

Ddd195, 2025-03-10:
> UMA-VI只有初始部分和结尾部分用SFM和标定板获取的真值，中间轨迹部分是没有真值的，所以在评估时不能对齐尺度，也不能用于评估带回环检测的算法。结果不稳定有可能是每次轨迹的微小改变会影响evo对齐的结果，造成了得到的RMSE差很多。另外，可以忽略最后的"terminate called without an active exception"

谢谢，请问UMAVI中间部分没有真值，只有初始和结尾部分，那ape能够反映算法在中间序列的鲁棒性吗

Ddd195, 2025-03-10:
> UMA-VI只有初始部分和结尾部分用SFM和标定板获取的真值，中间轨迹部分是没有真值的，所以在评估时不能对齐尺度，也不能用于评估带回环检测的算法。结果不稳定有可能是每次轨迹的微小改变会影响evo对齐的结果，造成了得到的RMSE差很多。另外，可以忽略最后的"terminate called without an active exception"

另外，我想我找到了不稳定的原因，我把局部地图优化选取的关键帧数量从9改为了11，后续结果的APE都十分稳定，csc1数据集RMSE均为0.8，是不是有其他参数需要调整以使得达到论文中的0.5效果呢

xukuanHIT, 2025-03-14:
可以在一定程度上反映。如果中间部分跟踪的不好，轨迹的两端和真值一般也会差很多