sair-lab/AirSLAM #187: Erron lightglue building Error in SuperPoint building

kinggreat24, 2025-02-07:
[ INFO] [1738941254.611129062]: rviz version 1.14.25
[ INFO] [1738941254.611220531]: compiled against Qt version 5.12.8
[ INFO] [1738941254.611227790]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1738941254.623850816]: Forcing OpenGl version 0.
config_file = /home/kinggreat24/air_slam_ws/src/AirSLAM-master/configs/visual_odometry/vo_oivio.yaml
config done
[02/07/2025-23:14:17] [I] [TRT] [MemUsageChange] Init CUDA: CPU +16, GPU +0, now: CPU 61, GPU 1130 (MiB)
[ INFO] [1738941258.033024747]: Stereo is NOT SUPPORTED
[ INFO] [1738941258.033139233]: OpenGL device: D3D12 (NVIDIA GeForce RTX 3080 Laptop GPU)
[ INFO] [1738941258.033175395]: OpenGl version: 3.1 (GLSL 1.4).
[02/07/2025-23:14:22] [I] [TRT] [MemUsageChange] Init builder kernel library: CPU +2176, GPU +406, now: CPU 2392, GPU 1536 (MiB)
Erron lightglue building
Error in SuperPoint building
[visual_odometry-2] process has finished cleanly
这是因为什么原因，能给点指导吗

kinggreat24, 2025-02-08:
1. CUDA的版本是12.1
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Feb__7_19:32:13_PST_2023
Cuda compilation tools, release 12.1, V12.1.66
Build cuda_12.1.r12.1/compiler.32415258_0 
2. TensorRT的版本是TensorRT-8.6.1.6 ，例子也能通过
[02/08/2025-16:01:57] [I] Output:
[02/08/2025-16:01:57] [I]  Prob 0  0.0000 Class 0:
[02/08/2025-16:01:57] [I]  Prob 1  0.0000 Class 1:
[02/08/2025-16:01:57] [I]  Prob 2  0.0000 Class 2:
[02/08/2025-16:01:57] [I]  Prob 3  0.0000 Class 3:
[02/08/2025-16:01:57] [I]  Prob 4  0.0000 Class 4:
[02/08/2025-16:01:57] [I]  Prob 5  0.0000 Class 5:
[02/08/2025-16:01:57] [I]  Prob 6  1.0000 Class 6: **********
[02/08/2025-16:01:57] [I]  Prob 7  0.0000 Class 7:
[02/08/2025-16:01:57] [I]  Prob 8  0.0000 Class 8:
[02/08/2025-16:01:57] [I]  Prob 9  0.0000 Class 9:
[02/08/2025-16:01:57] [I]
&&&& PASSED TensorRT.sample_onnx_mnist [TensorRT v8601] # ./sample_onnx_mnist

xukuanHIT, 2025-02-16:
你好，按照之前的一些经验，这一般是环境的问题。你可以试试把这[一行](https://github.com/sair-lab/AirSLAM/blob/19569d180e4b4b4e2aaa8381916fd599e62b882c/src/super_point.cpp#L16C45-L16C60)的kINTERNAL_ERROR改成kINFO，多输出一些信息。另外，你用docker试过吗？