sair-lab/AirSLAM #189: Running AirSLAM on the KITTI dataset

xcanqi, 2025-02-25:
你好，我尝试跑KITTI数据集，但是系统一直没有成功初始化：
good_stereo_point =0 
Not enough stereo points to initialize!
[image]

RVIZ也没有任何图像显示：
[image]

我已经参照issues#96修改了两个yaml文件：
（1）KITTI04-12.yaml
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

# original 1226*370 need to crop to 1224*368
image_width: 1224
image_height: 368
use_imu: 0

depth_lower_thr: 0.1
depth_upper_thr: 50.0
max_y_diff: 2

# Calibration
distortion_type: 0  # 0 for undistorted inputs, 1 for radial-tangential: [k1, k2, p1, p2, k3], 2 for equidistant/fisheye:  [k1, k2, k3, k4, 0].

# Camera calibration and distortion parameters (OpenCV) 
cam0:
  #intrinsics: [718.856, 718.856, 607.1928, 185.2157] # fx, fy, cx, cy
  intrinsics: [7.070912000000e+02, 7.070912000000e+02, 6.018873000000e+02, 1.831104000000e+02] # fx, fy, cx, cy
  distortion_coeffs: [0.0, 0.0, 0.0, 0.0, 0] # Distortion parameter
  T_type: 0           # 0 for Euroc format, the following T is Tbc. 1 for Kalibr format, the following T is Tcb
  T: # Convert camera to IMU
  - [1.0, 0.0, 0.0, 0.0]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
  - [0.0, 0.0, 0.0, 1.0]

cam1:
  intrinsics: [7.070912000000e+02, 7.070912000000e+02, 6.018873000000e+02, 1.831104000000e+02] 
  distortion_coeffs: [0.0, 0.0, 0.0, 0.0, 0]
  T_type: 0           # 0 for Euroc format, the following T is Tbc. 1 for Kalibr format, the following T is Tcb
  T: 
  - [1.0, 0.0, 0.0, 0.0]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]

（2）vo_kitti.yaml
plnet:
  use_superpoint: 1
  max_keypoints: 400
  keypoint_threshold: 0.004
  remove_borders: 4 
  line_threshold: 0.75
  line_length_threshold: 50

point_matcher:
  matcher: 0   # 0 for lightglue, 1 for superglue
  image_width: 1224
  image_height: 368
  onnx_file: "superpoint_lightglue.onnx"
  engine_file: "superpoint_lightglue.engine"

keyframe:
  min_init_stereo_feature: 30
  lost_num_match: 10
  min_num_match: 30
  max_num_match: 80
  tracking_point_rate: 0.65  
  tracking_parallax_rate: 0.1

optimization:
  tracking:
    mono_point: 50
    stereo_point: 75
    mono_line: 50
    stereo_line: 75
    rate: 0.5
  backend:
    mono_point: 50
    stereo_point: 75
    mono_line: 50
    stereo_line: 75
    rate: 0.5

ros_publisher:
  feature: 1
  feature_topic: "/AirSLAM/feature"
  frame_pose: 1
  frame_pose_topic: "/AirSLAM/frame_pose"
  frame_odometry_topic: "/AirSLAM/LatestOdometry"
  keyframe: 1
  keyframe_topic: "/AirSLAM/keyframe"
  path_topic: "/AirSLAM/odometry"
  map: 1
  map_topic: "/AirSLAM/map"
  mapline: 1
  mapline_topic: "/AirSLAM/mapline"
  reloc: 0
  reloc_topic: "/AirSLAM/reloc"

 此外，我还修改了部分代码：
（1）在visual_odometry.cpp中crop了数据集图像，裁剪为1224*368
（2）在dataset.cc中，image_dir改为对应KITTI数据集的名称
（3）在utils.cc中，修改了StringTimeToDouble(std::string time_str)函数：
double StringTimeToDouble(std::string time_str){
  // euroc
  // time_str.erase(std::remove_if(time_str.begin(), time_str.end(), [](char c) { return c == '.'; }), time_str.end());
  // double time_double1 = atof(time_str.substr(0, 10).c_str());
  // double time_double2 = atof(("0." + time_str.substr(10, time_str.length()+1)).c_str());
  // return (time_double1 + time_double2);

  // kitti
  double timestampMicroseconds = atof(time_str.c_str());
  // 将微秒转换为纳秒
  return timestampMicroseconds * 1000.0;
}

不仅如此，我也重新生成了所有.engine文件，但是仍然无法正常运行。请您协助我解决这个问题，谢谢！
我的环境为：Ubuntu20.04 cuda12.1 TensorRT8.6.1.6  OPENCV4.2 RTX 4060Ti (在EuRoC数据集能够正常运行并输出轨迹结果)

Zsudoku, 2025-02-27:
KITTI04-12.yaml 里的max_y_diff ，适当增加一下

xukuanHIT, 2025-02-27:
你好，可以上面的方法试着把max_y_diff增加一下。不过AirSLAM在KITTI上的效果应该不是很好。在用PLNet提取feature时，由于TensorRT对一些pytorch的算子支持的不是很好，我们把输入resize到了512*512，KITTI的长宽比例可能会导致resize后的图像失真比较严重