sair-lab/AirSLAM #196: 更换数据集运行发现实时性较差

tinal0629, 2025-03-19:
您好!想请教一下,为什么把您的airslam代码用于我自己的数据集上会出现数据处理速度过慢的问题呢(如下图),我对代码的改动是增加了rosbag的话题接收功能,然后取消了assert(std::abs(end_time-t0)<1e-5 || end_time<0)

[image]

robinnnnn1, 2025-03-20:
你好，请问你的这个问题解决了吗？

xukuanHIT, 2025-03-20:
你好，从提供的信息的来看，关键帧插入的比较频繁，关键帧耗时比普通帧要多很多，这可能造成了程序跑的比较慢。关键帧插入比较频繁可能因为：
- 图像帧率比较低或者运动速度过快，导致相邻图像之间的overlap比较小。
- 跟踪到的特征点太少。原因可能是相机标定问题，图像模糊问题等。

另外，程序跑的慢也可能因为：
- 图像尺寸比较大，使得PLNet特征检测比较慢
- 检测的特征点太多，使得Lightglue匹配比较耗时
- GPU被其它程序占用

你可以提供更多信息吗？包括：
- 你的软硬件平台
- 数据集图像尺寸，帧率等信息
- 相比于EuRoC慢多少？

tinal0629, 2025-03-24:
感谢您的回答，您提到的这个关键帧过多的问题，经过我修改参数tracking_point_rate和tracking_parallax_rate已经修正了。但是现在另外一个问题就是即使是普通帧，处理的依旧很慢，经过查看发现是在AddInput函数的这个语句wait了很长时间，于是把3改成了20，发现速度上没有太大提升。

`while (_data_buffer.size() > 20 && !_shutdown) {    usleep(2000);    wait_count++;}`

所以我想可能还是`ExtractFeatureThread`这个线程处理速度比较慢，应该是yaml参数设定的问题？但是我改了max_keypoints和keypoint_threshold发现速度有提升但是wait的时间还是较长，所以还是请您帮我看看该如何修改参数，或者帮忙分析一下问题可能出在哪里？

plnet:
use_superpoint: 1
max_keypoints: 200
keypoint_threshold: 0.05
remove_borders: 4
line_threshold: 0.75
line_length_threshold: 7

point_matcher:
matcher: 0   # 0 for lightglue, 1 for superglue
image_width: 640
image_height: 512
onnx_file: "superpoint_lightglue.onnx"
engine_file: "superpoint_lightglue.engine"

keyframe:
min_init_stereo_feature: 5
lost_num_match: 10
min_num_match: 5
max_num_match: 80
#tracking_point_rate: 0.65

tracking_point_rate: 0.30
#tracking_parallax_rate: 0.1
tracking_parallax_rate: 0.3
下图是我修改一些参数之后的运行截图：

[image]
（其中相邻关键帧匹配是这个函数运行时间：_point_matcher->MatchingPoints(features_last_keyframe, left_features, matches, true);normalframe detect left features是 _feature_detector->Detect(image_left_rect, left_features);
      frame->AddLeftFeatures(left_features, left_lines);的时间）

我的图像的帧率是 10 FPS，图像的尺寸是image_height: 512，image_width: 640。

CPU：Intel® Core™ Ultra 9 185H，GPU：RTX4060

下图是用您的示例跑的euroc处理时间：

[image]

tinal0629, 2025-03-24:
@xukuanHIT

xukuanHIT, 2025-03-25:
Wait时间比较长应该是被后面线程阻塞了。可以测测feature detection和优化部分的时间，或者先确定一下是那部分耗时比较长

tinal0629, 2025-03-27:
您好，我发现是跟踪线程调用的TrackFrame中调用的优化函数  int num_inliers = FramePoseOptimization(ref_frame, current_frame, matched_mappoints, inliers, _preinteration);耗时比较长，导致了线程阻塞，请问该如何解决呢

[image]
@xukuanHIT 期待您的回复

xukuanHIT, 2025-03-28:
你好，这个优化函数的速度不太合理，可以进入函数继续拆分，进一步定位是具体哪个步骤慢。目前来看这个很有可能是g2o的问题。你装的g2o哪个版本？是否用的提供的docker？有运行过g2o自带的测试用例吗？

tinal0629, 2025-03-29:
您好，我使用的是docker，刚刚删除容器后新建了容器这个优化函数慢的问题解决啦，实时性可以跟得上，感谢您的指导。
但是现在突然出现两个问题：

1. 在rviz中无法显示点云和路径信息，rostopic可以显示话题但是echo的话没有显示具体发布的信息

[image]

[image]

2.在车辆右转时会跟踪丢失，具体的报错如下：

[image]
有点不知道该怎么解决。@xukuanHIT 期待您的回复

xukuanHIT, 2025-04-02:
@tinal0629 你好，综合 #197 的情况，有可能是相机标定效果不好影响了特征匹配和优化

6615, 2025-04-24:
@tinal0629 您好，可以分享下您修改完rosbag的话题接收功能的代码吗？新手想练习复现下别的数据集尝试下，邮箱是 cyrushimself123@gmail.com,谢谢🌹

sun421-cloud, 2025-06-12:
@tinal0629 您好，可以分享下您修改完rosbag的话题接收功能的代码吗？新手想练习复现下别的数据集尝试下，邮箱是chengfeng_sun@163.com,谢谢🌹

cChrister, 2025-07-23:
@tinal0629  您好，可以分享下您修改完rosbag的话题接收功能的代码吗？新手想练习复现下别的数据集尝试下，邮箱是[1726233711@qq.com](1726233711@qq.com),谢谢🌹