sair-lab/AirSLAM #197: 关于地图尺度问题

Zsudoku, 2025-03-26:
作者您好，我在使用我自己的双目相机建图时，发现建出来的特征点线地图尺度很小，请问大概是哪里的问题呢，我没有使用imu。
camera.yaml
`image_height: 720
image_width: 1280
use_imu: 0

depth_lower_thr: 1
depth_upper_thr: 20.0
max_y_diff: 5

# Calibration
distortion_type: 1  # 0 for undistorted inputs, 1 for radial-tangential: [k1, k2, p1, p2, k3], 2 for equidistant/fisheye:  [k1, k2, k3, k4, 0].
cam0:
  intrinsics: [667.3003, 670.0276, 596.8533,395.0267] # fx, fy, cx, cy
  distortion_coeffs: [0.0895, -0.1429, 0.0029, -0.0001192, 0.0219]
  T_type: 0           # 0 for Euroc format, the following T is Tbc. 1 for Kalibr format, the following T is Tcb
  T: 
  - [1.0, 0.0, 0.0, 0.0]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
  - [0.0, 0.0, 0.0, 1.0]
cam1:
  intrinsics: [673.4658, 675.6429, 602.4980, 396.7246] # fx, fy, cx, cy
  distortion_coeffs: [0.1034, -0.189, 0.0024, -0.00063025, 0.0593]
  T_type: 0           
  T: 
  - [1.0, 0.0, 0.0, -0.12]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
  - [0.0, 0.0, 0.0, 1.0]`

vo_euroc.yaml
`plnet:
  use_superpoint: 1
  max_keypoints: 400
  keypoint_threshold: 0.004
  remove_borders: 4 
  line_threshold: 0.7
  line_length_threshold: 90

point_matcher:
  matcher: 0   # 0 for lightglue, 1 for superglue
  image_width: 1280
  image_height: 720
  onnx_file: "superpoint_lightglue.onnx"
  engine_file: "superpoint_lightglue.engine"

keyframe:
  min_init_stereo_feature: 15
  lost_num_match: 10
  min_num_match: 30
  max_num_match: 50
  tracking_point_rate: 0.7
  tracking_parallax_rate: 0.1

optimization:
  tracking:
    mono_point: 50
    stereo_point: 75
    mono_line: 50
    stereo_line: 75
    rate: 0.5
  backend:
    mono_point: 50
    stereo_point: 75
    mono_line: 50
    stereo_line: 75
    rate: 0.5

ros_publisher:
  feature: 1
  feature_topic: "/AirSLAM/feature"
  frame_pose: 1
  frame_pose_topic: "/AirSLAM/frame_pose"
  frame_odometry_topic: "/AirSLAM/LatestOdometry"
  keyframe: 1
  keyframe_topic: "/AirSLAM/keyframe"
  path_topic: "/AirSLAM/odometry"
  map: 1
  map_topic: "/AirSLAM/map"
  mapline: 1
  mapline_topic: "/AirSLAM/mapline"
  reloc: 0
  reloc_topic: "/AirSLAM/reloc"`

@xukuanHIT 

期待您的回复

tinal0629, 2025-03-26:
同面临这个尺度太小的问题，我用的是air_vo也是双目，换了数据集发现轨迹形状是出来了但是尺度误差较大。在我这里是配置文件的bf值如果按照基线*fx计算的话是260左右，写进去的话会出现只能提取绿色的特征点没有线特征而且无法跟踪（轨迹txt文件是空白）。后来误打误撞把bf改成大概15就可以跑出来形状但是尺度很小，不知道是为什么，请您指点一下 @xukuanHIT

xukuanHIT, 2025-03-26:
@Zsudoku @tinal0629 尺度问题可能是两个原因造成的：
1. 相机参数标定的不准确，比如fx, fy，双目基线和畸变参数。可以利用[这个方法](https://github.com/sair-lab/AirSLAM/issues/154#issuecomment-2451387128)检查一下双目校正的结果
2. 环境比较空旷，特征点距离比较远，使得双目三角化的精度变得很低。

xukuanHIT, 2025-03-28:
@tinal0629 校正效果很差。正常双目校正后相同位置应该在左右双目的同一高度上，可以观察你图片上的杆子，相同位置y坐标差很多。你可以拿EuRoC的参数和数据做个对比看下

Zsudoku, 2025-04-01:
感谢您的回复，很抱歉现在才给您最新的回复。这是我使用如下代码输出的结果：

[image]

请问这样的结果算是好结果吗？
```
#include <iostream>
#include <chrono>
#include <opencv2/opencv.hpp>
#include <Eigen/Core>
#include <ros/ros.h>
#include <thread>

#include "read_configs.h"
#include "dataset.h"
#include "map_builder.h"
#include "debug.h"

#include "plnet.h"
#include "feature_detector.h"

int main(int argc, char **argv) {
  ros::init(argc, argv, "air_slam");

  // std::string camera_config_path = "/media/code/ubuntu_files/airvio/catkin_ws/src/AirVIO/configs/camera/euroc.yaml";
  // std::string dataroot = "/media/data/datasets/euroc/seq/MH_01_easy/";
  // std::string model_dir = "/media/code/ubuntu_files/airvio/catkin_ws/src/AirVIO/output";
  // std::string save_root = "/media/code/ubuntu_files/airvio/catkin_ws/src/AirVIO/debug/point_detection";

  std::string camera_config_path = "/my_workspace/catkin_ws/src/AirSlam/configs/camera/euroc_720_2.yaml";
  std::string dataroot = "/my_workspace/catkin_ws/src/AirSlam/dataroot/0210_factory_2/cam0/data";
  std::string model_dir = "/my_workspace/catkin_ws/src/AirSlam/output";
  std::string save_root = "/my_workspace/catkin_ws/src/AirSlam/save";

  MakeDir(save_root);

  PLNetConfig plnet_config;
  plnet_config.use_superpoint = 1;
  plnet_config.max_keypoints = 400;
  plnet_config.keypoint_threshold = 0.04;
  plnet_config.remove_borders = 4;
  plnet_config.line_threshold = 0.5;
  plnet_config.line_length_threshold = 50;
  plnet_config.SetModelPath(model_dir);

  CameraPtr _camera = std::shared_ptr<Camera>(new Camera(camera_config_path));

  FeatureDetectorPtr feature_detector = std::shared_ptr<FeatureDetector>(new FeatureDetector(plnet_config));

  std::vector<std::string> image_names;
  GetFileNames(dataroot, image_names);
  size_t dataset_length = image_names.size();

  cv::Mat image = cv::imread("/my_workspace/catkin_ws/src/AirSlam/dataroot/0210_factory_2/cam0/data/17333604020700000000.jpg", 0);

  cv::Mat image_left_rect;
  _camera->UndistortImage(image, image_left_rect);
  cv::imwrite("/my_workspace/catkin_ws/src/AirSlam/save/left.jpg",image_left_rect);

  cv::Mat image_right = cv::imread("/my_workspace/catkin_ws/src/AirSlam/dataroot/0210_factory_2/cam1/data/17333604020700000000.jpg", 0);

  cv::Mat image_right_rect;
  _camera->UndistortImage(image_right, image_right_rect);
  cv::imwrite("/my_workspace/catkin_ws/src/AirSlam/save/right.jpg",image_right_rect);

 // 新功能：左右拼接图像，并在拼接图上每隔30个像素绘制一条水平直线
  cv::Mat combined_image;
  // 使用hconcat直接拼接左右图像
  cv::hconcat(image_left_rect, image_right_rect, combined_image);

// 如果拼接图像是灰度图，则转换为BGR彩色图像
if (combined_image.channels() == 1) {
    cv::cvtColor(combined_image, combined_image, cv::COLOR_GRAY2BGR);
}

for (int y = 0; y < combined_image.rows; y += 30) {
  cv::line(combined_image, cv::Point(0, y), cv::Point(combined_image.cols - 1, y),
           cv::Scalar(0, 255, 0), 1, cv::LINE_AA);
}

  // 保存拼接并绘制直线后的图像
  cv::imwrite("/my_workspace/catkin_ws/src/AirSlam/save/combined.jpg", combined_image);

  ros::shutdown();

  return 0;
}
```
@xukuanHIT  期待您的回复

Zsudoku, 2025-04-03:
@xukuanHIT  您好！首先非常感谢您的回复。我这里还有另一个问题向您请教，如何将用于定位的特征点云图等比例升级为稠密点云图，非常期待您的回复！

xukuanHIT, 2025-04-14:
@Zsudoku 你好，抱歉这么晚回复。将点云稠密化的工作我们没有尝试过，可能无法提供一些有效的建议

Zsudoku, 2025-04-16:
> [@Zsudoku](https://github.com/Zsudoku) 你好，抱歉这么晚回复。将点云稠密化的工作我们没有尝试过，可能无法提供一些有效的建议

好的谢谢！

closed 2025-04-16 by Zsudoku