sair-lab/AirSLAM #201: The error occurs when running catkin_make with the message "The compiler cannot recognize the types TempfileControlFlags and HardwareCompatibilityLevel.

madong19910929, 2025-03-31:
Airslam/src/AirSLAM/3rdparty/tensorrtbuffer/include/sample_options.h:182:25: error: ‘TempfileControlFlags’ in namespace ‘nvinfer1’ does not name a type
  182 |     constexpr nvinfer1::TempfileControlFlags getTempfileControlDefaults()
      |                         ^~~~~~~~~~~~~~~~~~~~
/home/slam/Airslam/src/AirSLAM/3rdparty/tensorrtbuffer/include/sample_options.h:238:19: error: ‘HardwareCompatibilityLevel’ in namespace ‘nvinfer1’ does not name a type
  238 |         nvinfer1::HardwareCompatibilityLevel hardwareCompatibilityLevel{nvinfer1::HardwareCompatibilityLevel::kNONE};
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~

The error occurs when running catkin_make with the message "The compiler cannot recognize the types TempfileControlFlags and HardwareCompatibilityLevel. How should this be resolved? Thank you."
I am using TensorRT version 8.4.1.5.

wsy2564593234, 2025-04-09:
Excuse me, has the problem been solved yet?

xukuanHIT, 2025-04-11:
@madong19910929 Hi, it’s probably an issue with the development environment. Are you using the provided Docker environment?

closed 2025-05-09 by madong19910929