sair-lab/AirSLAM #203: Error during conversion of plnet1 ONNX to TensorRT engine – ForeignNode (/Cast_1.../Gather_69) implementation not found

SongSiWoon, 2025-04-08:
Hello,

I'm encountering an error while converting the plnet1 ONNX model to a TensorRT engine on my Jetson Orin Nano (JetPack 6.2) environment using Docker. The Docker container is based on the image nvcr.io/nvidia/l4t-tensorrt:r8.6.2-devel (CUDA 12.2, TensorRT 8.6.2). The issue does not occur when running the model normally; the error shows up only when I attempt to build the engine separately with the plnet1 ONNX model.

I used the following command to build the engine:

```
/usr/src/tensorrt/bin/trtexec \
  --onnx=/workspace/src/AirSLAM/output/plnet_s1.onnx \
  --saveEngine=/workspace/src/AirSLAM/output/plnet_s1.engine \
  --explicitBatch \
  --minShapes=juncs_pred:1x2,lines_pred:1x4,idx_lines_for_junctions:1x2,inverse:1x1,iskeep_index:1x1,loi_features:1x16x16x16,loi_features_thin:1x4x16x16,loi_features_aux:1x4x16x16 \
  --optShapes=juncs_pred:250x2,lines_pred:20000x4,idx_lines_for_junctions:20000x2,inverse:20000x1,iskeep_index:20000x1,loi_features:1x128x128x128,loi_features_thin:1x4x128x128,loi_features_aux:1x4x128x128 \
  --maxShapes=juncs_pred:500x2,lines_pred:50000x4,idx_lines_for_junctions:50000x2,inverse:50000x1,iskeep_index:50000x1,loi_features:1x512x512x512,loi_features_thin:1x4x512x512,loi_features_aux:1x4x512x512 \
  --fp16
```
During the engine build process, I observed the following warnings and errors:

Warnings:

- The --explicitBatch flag is deprecated and has no effect.

- A warning indicating that the ONNX model was generated with INT64 weights, with a message like “Your ONNX model has been generated with INT64 weights… Attempting to cast down to INT32,” and that some values have been clamped.

Critical Error:

```
value.h:682: DCHECK(new_vv.cast(myelinTypeUint64)) failed.
...
[E] Error[10]: Could not find any implementation for node {ForeignNode[/Cast_1.../Gather_69]}.
[E] Error[10]: [optimizer.cpp::computeCosts::3869] Error Code 10: Internal Error (Could not find any implementation for node {ForeignNode[/Cast_1.../Gather_69]}).
[E] Engine could not be created from network
[E] Building engine failed
```
My Questions:

1. Root Cause:

- Is the error caused by the ONNX model being generated with INT64 weights that TensorRT does not natively support (thus automatically casting them to INT32), leading to potential issues during the conversion?

- Or does this error indicate that certain operations in the ONNX model (such as those involving Cast or Gather) are not currently implemented in TensorRT 8.6.2?

2. Potential Solutions:

- Should the ONNX export process be modified to use supported types (e.g., using INT32 instead of INT64) to avoid these issues?

- Would a custom plugin implementation or modifications to the model be required to handle the problematic operations?

- Are there any known workarounds or updates to TensorRT that address this error?

Any insights or suggestions for resolving this issue would be greatly appreciated.

Thank you!

ynma-hanvo, 2025-05-13:
i am get : 
10: [optimizer.cpp::computeCosts::3728] Error Code 10: Internal Error (Could not find any implementation for node {ForeignNode[/Equal.../backbone/point_detector/Unsqueeze_10]}.)
for building the  plnet_s0.onnx file.

any suggestion?

ynma-hanvo, 2025-05-15:
thanks. i solved the problem by increase the workspace size to 4G   ;

rendebin, 2025-06-11:
> thanks. i solved the problem by increase the workspace size to 4G ;

Could you please provide the code for exporting models to ONNX? I would be very grateful.

shin-ka, 2025-07-14:
@SongSiWoon were you able to resolve this error?

SongSiWoon, 2025-07-15:
Hi @shin-ka

Yes, I was able to resolve the issue by adjusting the shape options in `trtexec`.  
It seems that `maxShapes` like `lines_pred:50000x4` and `loi_features:1x512x512x512` caused internal Myelin errors during engine build.

I reduced them slightly (e.g., `lines_pred:40000x4`, `loi_features:1x256x256x256`) and the build succeeded.

I originally tested the fix using `trtexec`, but if you're building the engine from code,  
you can directly modify the shape settings in [`plnet.cpp`](https://github.com/sair-lab/AirSLAM/blob/master/src/plnet.cpp#L130-L154).  
That section defines the `profile_stage2->setDimensions(...)` calls, and you can apply the same shape values there instead.

For reference, here’s the exact `trtexec` CLI command I used that worked:

```bash
/usr/src/tensorrt/bin/trtexec \
  --onnx=AirSLAM/output/plnet_s1.onnx \
  --saveEngine=AirSLAM/output/plnet_s1.engine \
  --fp16 \
  --skipInference \
  --verbose \
  --builderOptimizationLevel=1 \
  --workspace=2048 \
  --minShapes=juncs_pred:1x2,lines_pred:1x4,idx_lines_for_junctions:1x2,inverse:1x1,iskeep_index:1x1,loi_features:1x16x16x16,loi_features_thin:1x4x16x16,loi_features_aux:1x4x16x16 \
  --optShapes=juncs_pred:250x2,lines_pred:20000x4,idx_lines_for_junctions:20000x2,inverse:20000x1,iskeep_index:20000x1,loi_features:1x128x128x128,loi_features_thin:1x4x128x128,loi_features_aux:1x4x128x128 \
  --maxShapes=juncs_pred:500x2,lines_pred:50000x4,idx_lines_for_junctions:40000x2,inverse:40000x1,iskeep_index:40000x1,loi_features:1x256x256x256,loi_features_thin:1x4x512x512,loi_features_aux:1x4x512x512

shin-ka, 2025-07-16:
@SongSiWoon Thanks a lot for the quick and clear response!
I was able to solve the issue following your steps!

shin-ka, 2025-07-17:
Hi @rendebin, I didn't export plnet_s0 and plnet_s1 to ONNX myself, I used the ones provided by the author in this repo
[plnet_s0.onnx](https://github.com/sair-lab/AirSLAM/blob/master/output/plnet_s0.onnx) and [plnet_s.onnx](https://github.com/sair-lab/AirSLAM/blob/master/output/plnet_s1.onnx)