sair-lab/AirSLAM #208: 运行 vo_euroc.launch时中断

wsy2564593234, 2025-04-11:
以下是终端提示的内容：
... logging to /home/eric/.ros/log/1316f3a8-16b4-11f0-a2b5-50ebf65be2a1/roslaunch-wsy-13750.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://wsy:38589/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /visual_odometry/camera_config_path: /home/eric/view_w...
 * /visual_odometry/config_path: /home/eric/view_w...
 * /visual_odometry/dataroot: /home/eric/datase...
 * /visual_odometry/model_dir: /home/eric/view_w...
 * /visual_odometry/saving_dir: /home/eric/view_w...

NODES
  /
    rviz (rviz/rviz)
    visual_odometry (air_slam/visual_odometry)

auto-starting new master
process[master]: started with pid [13779]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1316f3a8-16b4-11f0-a2b5-50ebf65be2a1
process[rosout-1]: started with pid [13822]
started core service [/rosout]
process[visual_odometry-2]: started with pid [13825]
process[rviz-3]: started with pid [13826]
config_file = /home/eric/view_work/AirSLAM/src/AirSLAM/configs/visual_odometry/vo_euroc.yaml
config done
inotify_add_watch("/home/eric/.config/fcitx/dbus") failed: "No space left on device"
inotify_add_watch("/home/eric/.config/fcitx/dbus/edc7c66fb8f644b49c7a9e6cfdb90896-1") failed: "No space left on device"
[ INFO] [1744362312.347232342]: rviz version 1.13.30
[ INFO] [1744362312.347264752]: compiled against Qt version 5.9.5
[ INFO] [1744362312.347272723]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1744362312.352013139]: Forcing OpenGl version 0.
map_builder done
[ INFO] [1744362312.663236616]: Stereo is NOT SUPPORTED
[ INFO] [1744362312.663269106]: OpenGL device: NVIDIA GeForce RTX 3060/PCIe/SSE2
[ INFO] [1744362312.663277571]: OpenGl version: 4.6 (GLSL 4.6).
dataset done
i ====== 0
One Frame Processinh Time: 0 ms.
i ====== 1
One Frame Processinh Time: 0 ms.
i ====== 2
One Frame Processinh Time: 0 ms.
i ====== 3
One Frame Processinh Time: 0 ms.
i ====== 4
One Frame Processinh Time: 0 ms.
i ====== 5
Initialization is done!
One Frame Processinh Time: 479 ms.
i ====== 6
One Frame Processinh Time: 17 ms.
i ====== 7
One Frame Processinh Time: 3 ms.
i ====== 8
One Frame Processinh Time: 4 ms.
i ====== 9
One Frame Processinh Time: 63 ms.
i ====== 10
One Frame Processinh Time: 5 ms.
i ====== 11
insert keyframe, id = 4
*** stack smashing detected ***: <unknown> terminated
[visual_odometry-2] process has died [pid 13825, exit code -6, cmd /home/eric/view_work/AirSLAM/devel/lib/air_slam/visual_odometry __name:=visual_odometry __log:=/home/eric/.ros/log/1316f3a8-16b4-11f0-a2b5-50ebf65be2a1/visual_odometry-2.log].
log file: /home/eric/.ros/log/1316f3a8-16b4-11f0-a2b5-50ebf65be2a1/visual_odometry-2*.log

xukuanHIT, 2025-04-11:
你好，我们之前没遇到过这个问题。是每次运行都会出现吗？

wsy2564593234, 2025-04-11:
感谢回复，是的，每次跑都是同样的情况。

wsy2564593234, 2025-04-11:
问题发生在src/AirSLAM/src/map_builder.cc中，第484行“  _map->InsertKeyframe(frame); ”处;
进入InsertKeyframe()（src/AirSLAM/src/map.cc中）后，程序中断在第114行“    LocalMapOptimization(frame);”处;
进入 LocalMapOptimization()（src/AirSLAM/src/map.cc中）后，程序中断在第755行“LocalmapOptimization(poses, points, lines, velocities, biases, camera_list, mono_point_constraints, 
      stereo_point_constraints, mono_line_constraints, stereo_line_constraints, imu_constraints, Rwg, _backend_optimization_config);”处;
进入LocalmapOptimization()（src/AirSLAM/src/g2o_optimization/g2o_optimization.cc中）后，程序中断在第347行“  optimizer.initializeOptimization();”处;

wsy2564593234, 2025-04-11:
目前rviz中只能出来[

[image]

](url)
只有一个关键帧以后就中断了。

xukuanHIT, 2025-04-11:
可以把[这几行](https://github.com/sair-lab/AirSLAM/blob/2b825166b05b351fc708c70cee3fc8b5a5b380e9/src/g2o_optimization/g2o_optimization.cc#L85-L96)反注释，看一下输出结果。同时也看一下CPU和内存占用

wsy2564593234, 2025-04-14:
---------LocalmapOptimization----------
poses.size = 2
points.size = 278
lines.size = 10
velocities.size = 0
biases.size = 0
mono_point_constraints.size = 27
stereo_point_constraints.size = 409
mono_line_constraints.size = 4
stereo_line_constraints.size = 9
imu_constraints.size = 0
------------------------------------
内存和CPU都正常，没有跑满

xukuanHIT, 2025-04-14:
报错的地方是调用的g2o的函数，你使用的是g2o的哪个版本呢？g2o自带的demo你能运行成功吗？

wsy2564593234, 2025-04-15:
G2o我用的就是推荐版本，现在出现了一些别的问题，在网络反序列化是发生了段错误。
light_glue.cpp中，bool SuperPointLightGlue::deserialize_engine()函数执行到nvinfer1::IRuntime *runtime = nvinfer1::createInferRuntime(tensorrt_log::gLogger);时发生段错误。具体的报错信息如下：

=================================================================
==29611==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7fdf4cbba3fb bp 0x7fff12f78460 sp 0x7fff12f78090 T0)
==29611==The signal is caused by a READ memory access.
==29611==Hint: address points to the zero page.
    #0 0x7fdf4cbba3fa in createInferRuntime_INTERNAL (/usr/local/lib/libnvinfer.so.8+0x143b3fa)
    #1 0x7fdf5d673adc in createInferRuntime /home/eric/tensorrt/TensorRT-8.6.1.6/include/NvInferRuntime.h:3756
    #2 0x7fdf5d673adc in SuperPointLightGlue::deserialize_engine() /home/eric/view_work/AirSLAM/src/AirSLAM/src/light_glue.cpp:353
    #3 0x7fdf5d675206 in SuperPointLightGlue::build() /home/eric/view_work/AirSLAM/src/AirSLAM/src/light_glue.cpp:21
    #4 0x7fdf5d7ac34a in PointMatcher::PointMatcher(PointMatcherConfig const&) /home/eric/view_work/AirSLAM/src/AirSLAM/src/point_matcher.cc:16
    #5 0x7fdf5d956912 in MapBuilder::MapBuilder(VisualOdometryConfigs&, ros::NodeHandle) /home/eric/view_work/AirSLAM/src/AirSLAM/src/map_builder.cc:28
    #6 0x5580932b4d44 in main /home/eric/view_work/AirSLAM/src/AirSLAM/demo/visual_odometry.cpp:26
    #7 0x7fdf5a6e8c86 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21c86)
    #8 0x5580932b7e69 in _start (/home/eric/view_work/AirSLAM/devel/lib/air_slam/visual_odometry+0x13e69)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/usr/local/lib/libnvinfer.so.8+0x143b3fa) in createInferRuntime_INTERNAL
==29611==ABORTING
Engine是能加载的，我也用TensorRT-8.6.1.6试加载了Engine文件，也可以加载成功。