sair-lab/AirSLAM #223: 关于AirSLAM论文中TarTanAir数据集的问题

zzy-2021, 2025-07-14:
作者您好！

AirSLAM是非常优秀的工作，感谢开源！
本人在尝试复现TarTanAir数据集的结果遇到一些问题：
1. 根据论文的说明，针对TarTanAir数据集，mapping时使用的是abandoned_factory中的序列，relocalization时是使用abandoned_factory_night中的序列。然而，我这样做发现relocalization的结果和论文中的不一致，我怀疑是我选择序列P000-P011的方法和论文不一致。
2. 我尝试选择序列以获得论文一致的结果时，发现abandoned_factory中的序列存在Easy和Hard两类，abandoned_factory_night的序列也存在Easy和Hard两类，也让序列选择更加困难，始终无法获得论文的实验结果。
3. 请问作者方便给出关于TarTanAir数据集各序列的详细信息吗？ 或者方便开源论文中预先选择的TarTanAir数据集吗？ 或者方便给出相关的代码来选择序列？ 非常感谢！！

xukuanHIT, 2025-07-15:
@zzy-2021 你好，建图序列是abandoned_factory的hard序列，重定位序列是挑选了和建图序列的图片距离在3m以内，相对角度在15°的图片（原文Section VII.D.1有介绍）。我们把重定位序列上传到了[这里](https://entuedu-my.sharepoint.com/:u:/g/personal/kuan_xu_staff_main_ntu_edu_sg/EXsPkQmuaKlEqpRAx6XqgL4BynHANDm331m5jgOfp3fSlw?e=rHje1r)。

zzy-2021, 2025-07-28:
好的，非常感谢！

closed 2025-08-08 by xukuanHIT